emqx/neuron #406: S7 support  2.0.0-rc1

jld-adriano, 2022-04-06:
It's documented in [Neuron docs](https://neugates.io/docs/en/latest/neuron-driver.html#s7comm) that you have support for the s7 protocol, but I can't seem to find the code implementation, can someone point me to the correct location?

lixiumei123, 2022-04-07:
Sorry, the s7 protocol is only supported in the commercial version and  isn't open source at this time.
Please download the commercial version:[Neuron V2.0.0-rc.1](https://www.emqx.com/zh/try?product=neuron)

lixiumei123, 2022-04-07:
> It's documented in [Neuron docs](https://neugates.io/docs/en/latest/neuron-driver.html#s7comm) that you have support for the s7 protocol, but I can't seem to find the code implementation, can someone point me to the correct location?

Sorry, the s7 protocol is only supported in the commercial version and  isn't open source at this time.
Please download the commercial version:[Neuron V2.0.0-rc.1](https://www.emqx.com/zh/try?product=neuron)

jld-adriano, 2022-04-07:
Got it, thank you!

closed 2022-04-07 by jld-adriano