emqx/neuron #463: 昨天下载的最新版本 web是404

xyfabc, 2022-05-06:
[image]

gc87, 2022-05-06:
The release version of this repository does not provide the front-end package, please go to the [official website](https://neugates.io/) to download the full installation package.

lixiumei123, 2022-05-07:
Front-end interface installation steps：
* Download the tar.gz package for Linux from the official website：
`$ wget https://www.emqx.com/zh/downloads/neuron/2.0.1/neuron-2.0.1-linux-amd64.tar.gz`
* Unpack the tar.gz package:
`$ tar -zxvf neuron-2.0.1-linux-amd64.tar.gz`
* Go to the extracted folder:
`$ cd neuron-2.0.1-linux`
* Copy the dist folder to the bottom of the neuron/build
`$ cp -r dist ${path}/neuron/build`

_notes:_ Choice of CPU architecture

fengzeroz, 2022-05-10:
We are aready considering including dashboard in release.

closed 2022-06-29 by fengzeroz