emqx/neuron #469: should apply License

icoco, 2022-05-07:
2.License

在未上传 License 时，Neuron是无法读写和上报数据的，请先在界面申请 License，如下图所示，在页面右上角，在关于下拉框中选择 License，进入到 License 界面。

icoco, 2022-05-08:
<div class="card-hd-with-btn" data-v-2c134c01=""><h3 class="card-title" data-v-2c134c01="">License</h3></div><div class="license" data-v-2c134c01=""><div class="el-descriptions emqx-descriptions" data-v-2c134c01=""><div class="el-descriptions__body">

License 类型retail
--
License 状态生效中
生效日期2022-03-25 09:36:12
失效日期2022-07-22 09:36:12
节点数限制100
节点下 Tag 数目限制10000
可用的 Pluginmodbus-plus-tcp, modbus-rtu, opcua, s7comm, fins, QnA3E, IEC60870-5-104

</div></div><div class="msg" data-v-2c134c01=""><div class="method-text" data-v-2c134c01=""><p data-v-2c134c01="">证书获取方式</p><p data-v-2c134c01=""> 1. <span class="payload-desc" data-v-2c134c01="">免费<a href="https://www.emqx.com/zh/apply-licenses/neuron" target="_blank" data-v-2c134c01="">申请</a> License</span></p><p data-v-2c134c01=""> 2. <span class="payload-desc" data-v-2c134c01=""><a href="https://www.emqx.com/zh/contact?product=neuron" target="_blank" data-v-2c134c01="">联系我们</a>，购买商业 License</span></p></div></div></div>License
License 类型retail
License 状态生效中
生效日期2022-03-25 09:36:12
失效日期2022-07-22 09:36:12
节点数限制100
节点下 Tag 数目限制10000
可用的 Pluginmodbus-plus-tcp, modbus-rtu, opcua, s7comm, fins, QnA3E, IEC60870-5-104

证书获取方式

1. 免费[申请](https://www.emqx.com/zh/apply-licenses/neuron) License

2. [联系我们](https://www.emqx.com/zh/contact?product=neuron)，购买商业 License

必须购买License 才可以用？

fengzeroz, 2022-05-09:
> ### License
> ## License 类型retail
> License 状态生效中 生效日期2022-03-25 09:36:12 失效日期2022-07-22 09:36:12 节点数限制100 节点下 Tag 数目限制10000 可用的 Pluginmodbus-plus-tcp, modbus-rtu, opcua, s7comm, fins, QnA3E, IEC60870-5-104
> 
> 证书获取方式
> 
> 1. 免费[申请](https://www.emqx.com/zh/apply-licenses/neuron) License
> 
> 2. [联系我们](https://www.emqx.com/zh/contact?product=neuron)，购买商业 License
> 
> License License 类型retail License 状态生效中 生效日期2022-03-25 09:36:12 失效日期2022-07-22 09:36:12 节点数限制100 节点下 Tag 数目限制10000 可用的 Pluginmodbus-plus-tcp, modbus-rtu, opcua, s7comm, fins, QnA3E, IEC60870-5-104
> 证书获取方式
> 
> 1. 免费[申请](https://www.emqx.com/zh/apply-licenses/neuron) License
> 2. [联系我们](https://www.emqx.com/zh/contact?product=neuron)，购买商业 License
> 
> 必须购买License 才可以用？

Open source version does not require a license.

closed 2022-06-29 by fengzeroz