emqx/neuron #472: Auto-generated source archives do not include submodules

hobbes1069, 2022-05-09:
For those that want to build from source from archives it would be nice to include source archives with the submodules included. I'm working on packaging this for Fedora linux and need complete archives.

hobbes1069, 2022-05-10:
I'm not sure if you're familiar with RPM spec files but I have worked around it for now by creating git archives for each submodule.
```
Source0:        https://github.com/%{owner}/%{project}/archive/refs/tags/%{version}/%{name}-%{version}.tar.gz
# Deal with git submodules not in the archive
Source1:        libcsptr.tar.gz
Source2:        uthash.tar.gz
Source3:        vector.tar.gz
[...]
autosetup -n %{project}-%{version}
gzip -dc %{SOURCE1} | tar -xf - -C extern/libcsptr
gzip -dc %{SOURCE2} | tar -xf - -C extern/uthash
gzip -dc %{SOURCE3} | tar -xf - -C extern/vector
```

fengzeroz, 2022-05-10:
> For those that want to build from source from archives it would be nice to include source archives with the submodules included. I'm working on packaging this for Fedora linux and need complete archives.

A GitHub tarball archive never includes the submodules, to my knownledge. We will consider creating source releases, which would be inclusive of all submodules.

hobbes1069, 2022-05-10:
Yes, not the first project I've run into this on, unfortunately it makes the auto-generated sources quite useless.

PR #1044 (2023-02-01): Make source self contained by removing uthash submodule
Close #472
closed 2023-02-01 by fengzeroz