emqx/neuron #474: error: implicit conversion from 'enum MQTTErrors' to 'neu_err_code_e'

hobbes1069, 2022-05-10:
```
/builddir/build/BUILD/neuron-2.0.1/src/connection/mqtt_c_client.c:809:11: error: implicit conversion from 'enum MQTTErrors' to 'neu_err_code_e' [-Werror=enum-conversion]
  809 |     error = mqtt_publish(&client->mqtt, topic, payload, len, qos);
      |           ^
```

Fedora Rawhide x86_64
GCC 12.0.1

gc87, 2022-05-10:
This compilation error is not triggered in lower versions of gcc, as a temporary solution, you can delete the -Werror compilation option in CMakeLists.txt, I will update my compilation environment to fix it.

https://github.com/emqx/neuron/blob/4b549cb287100c553fa9cd8e251837c04969a981/CMakeLists.txt#L16

hobbes1069, 2022-05-10:
Yes, already did. Fedora is good at catching these since it's always running the latest GCC. 

I hate to submit everything as a ticket, so is there a forum available? I'm having a linking issue with mqtt-c.

gc87, 2022-05-10:
> I hate to submit everything as a ticket, so is there a forum available? I'm having a linking issue with mqtt-c.

Yes, you can start a discussion [here](https://github.com/emqx/neuron/discussions).