emqx/neuron #477: relocation R_X86_64_PC32

wlq965, 2022-05-10:
/usr/bin/ld: /usr/local/lib/libmqttc.a(mqtt.c.o): relocation R_X86_64_PC32 against symbol `mqtt_fixed_header_rules' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: 最后的链结失败: bad value
collect2: error: ld returned 1 exit status

hobbes1069, 2022-05-10:
I ran into the same problem. This should really be reported to the MQTT-C project to turn on PIC by default.

gc87, 2022-05-11:
When compiling libmqtt-c deps, you need to turn on the -fPIC option. To do this, you can add `set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}" -fPIC)` in CMakeLists.txt of mqtt-c project or add `-DCMAKE_POSITION_INDEPENDENT_CODE=ON` in camke parameters.

gc87, 2022-06-29:
MQTT-C has enabled -fPIC compilation by default.
https://github.com/LiamBindle/MQTT-C/pull/166

closed 2022-06-29 by gc87