emqx/neuron #533: tests seem to be running during the cmake --build process

joey8869, 2022-06-13:
I'm attempting to package Neuron for Fedora in my COPR[1]. I've got it building but for some reason the tests seem to be running during the cmake --build process and not waiting for me to run ctest. This is causing the tests to run before the build process is 100% complete and frequently timing out on aarch64 builds which are far more resource constrained than other arches. Any pointers?
CMake Error at /usr/share/cmake/Modules/GoogleTestAddTests.cmake:83 (message):
  Error running test executable.

    Path: '/builddir/build/BUILD/neuron-2.0.1/redhat-linux-build/tests/json_persistence_test'
    Result: Process terminated due to timeout
    Output:
      Running main() from /builddir/build/BUILD/googletest-release-1.11.0/googletest/src/gtest_main.cc
      JsonAdapter.
        AdapterPersistenceDecode
      JsonPlugin.
        PluginPersistenceDecode
      JsonDatatags.
        DatatagPersistenceDecode
        DatatagPersistenceEncode
      JsonSubs.
        SubsPersistenceDecode
      JsonGroupConfigs.
        GroupConfigsPersistenceDecode
      JsonAdapterTest.
        AdapterPersistenceEncode
      JsonPluginTest.
        PluginPersistenceEncode
      JsonSubscriptionsTest.
        SbuscriptionsPersistenceEncode
      JsonGroupConfigTest.
        GroupConfigPersistenceEncode
      

Call Stack (most recent call first):
  /usr/share/cmake/Modules/GoogleTestAddTests.cmake:179 (gtest_discover_tests_impl)
[1] https://copr.fedorainfracloud.org/coprs/hobbes1069/IIoT/build/4508992/

From @hobbes1069

fengzeroz, 2022-06-13:
The DISABLE_UT parameter can be used to turn off build UT.

hobbes1069, 2022-06-13:
That's what I ended up doing but that's not a full fix. Fedora prefers that any type of unit testing be run if available. It should just be run in %check instead of %build.

fengzeroz, 2022-06-14:
Try to remove enable_testing() in cmake

closed 2023-01-17 by fengzeroz