emqx/neuron #634: neuron2.0.1 dashborad south refresh error

chq24707, 2022-07-12:
System: XuanTie 21.06 (kaitai)
Architecture: armv7l
Description: In the test of 3000 points, the acquisition frequency is 100ms. When the number of drives is too large, resulting in the error of the South refresh of the front end of the neuron, for example: 100 drives, 30 points under each drive, and the acquisition frequency is 100ms, the following errors will appear:
[image]
[image]

CPU and memory usage：
[image]

fengzeroz, 2022-07-21:
The default buf size of nng is small, and an error will be reported directly when there are many concurrent requests.

fengzeroz, 2022-07-21:
#653

closed 2022-07-21 by fengzeroz