emqx/neuron #651: 开源版neuron读取不到点位数据

chengbaifangyun123, 2022-07-18:
[image]
开源版neuron读取不到点位数据

chengbaifangyun123, 2022-07-22:
[image]
[image]

chengbaifangyun123, 2022-07-22:
是因为南向设备这个link状态的值不对，没连成功才会导致读取不到这个南向设备的数据

lixiumei123, 2022-07-22:
It's not that the value of link is wrong, is your localhost in the same segment as the modbus simulator's ip?

lixiumei123, 2022-07-22:
The open source version doesn't support such a configuration.
[image]

closed 2022-07-22 by lixiumei123