emqx/neuron #728: The web page keep loading when I apply another north application(ekuiper)

andyrewOO, 2022-08-23:
context：
 1. 4 south application；
 2. 2 north devices；
 3. 2 data streaming rules；

# ./neuron --version
Neuron 2.1.0 (31dbfb1 2022-06-30)
 
Neuron can have only one data streaming, default name：neuronStream.
when I new a south application(plugin select ekuiper),and apply this application. the  page no responding，and other page always keep loading.

https://user-images.githubusercontent.com/32356701/186088021-3dfc6a01-3792-42d8-a0df-5e0edd9bc02c.mp4

eeff, 2022-08-23:
To be clear, `ekuiper` is a north bound plugin. And the stream `neuronStream` is a concept more related to `LF-Edge Kuiper` than `Neuron` itself.

The current implementation of the `ekuiper` plugin uses the `pair0` protocol from the `NNG` library which allows only one-to-one communication between a server and a client. And within the context of `Neuron`, the `ekuiper` plugin acts as the server while a local running instance of `kuiperd` acts as the client.  The behavior of  `Neuron` version 2.1.0 that did not stop you from creating multiple nodes of the `ekuiper` plugin should be considered a flaw.

Please update the code or pull the docker images you used, as there are fixes.

eeff, 2022-12-01:
@fengzeroz I think this issue has been fixed in version 2.2

closed 2022-12-01 by fengzeroz