emqx/neuron #820: Does the opc shift component provide a trial？

iqihao, 2022-09-28:
Does the opcshift component provide a trial？

gc87, 2022-09-29:
Yes, but the official download link has not been provided yet, I uploaded it in the attachment of this issue.
[opcshift.zip](https://github.com/emqx/neuron/files/9670527/opcshift.zip)
It might be helpful to read [here](https://neugates.io/docs/en/latest/module-plugins/module-driver.html#opcda)

iqihao, 2022-09-29:
> Yes, but the official download link has not been provided yet, I uploaded it in the attachment of this issue. [opcshift.zip](https://github.com/emqx/neuron/files/9670527/opcshift.zip) It might be helpful to read [here](https://neugates.io/docs/en/latest/module-plugins/module-driver.html#opcda)

thanks

closed 2022-09-29 by iqihao