emqx/neuron #950: How to run it in my pi?

zhouzaihang, 2022-12-01:
Why did you remove the ci of package deb?
Or could you share your Dockerfile in emqx/neuron(Docker hub)?

eeff, 2022-12-01:
The repo has a script `package-sdk.sh` which could be used to produce a tarbar, you may have a look.

zhouzaihang, 2022-12-01:
> The repo has a script `package-sdk.sh` which could be used to produce a tarbar, you may have a look.

Not package sdk, and we want to package deb or rmp.
And I remember the repo had script in folds of packages.

eeff, 2022-12-01:
> > The repo has a script `package-sdk.sh` which could be used to produce a tarbar, you may have a look.
> 
> Not package sdk, and we want to package deb or rmp. And I remember the repo had script in folds of packages.

If you make a tarbar using `package-sdk.sh` then it's easy to build your own docker images.
As for deb/rpm packages, you may consult the deb/rpm documentaion. And if you interested, run ` git log -p package ` to see the scripts.

zhouzaihang, 2022-12-01:
ok

closed 2022-12-01 by zhouzaihang