emqx/neuron #1015: [mqtt][malloc][必现]amd64环境下，频繁publish某个topic，出现双链表异常的coredump

Jimmy-Wen, 2023-01-03:
#### mqtt topic
`/v1/nari/service/command
`
#### mqtt payload
`{
	"mid":10000000000000000,
	"type":"CMD_SERVICE",
	"deviceId":"sddddd4444444444444444444444",
	"timestamp":3743743749849844,
	"expire":-1,
        "app":"rtutest1",
        "param":{
            "cmd":"Tele-Adjusting",
            "paras":{
                 "tag300":22290,
                 "tag301":662.3,
		 "tag302":675.3
              }
           }
}
`

#### gdb调用栈
``[New Thread 0x7fffd57fa700 (LWP 80118)]
[New Thread 0x7fffd4ff9700 (LWP 80119)]

malloc(): unsorted double linked list corrupted

Thread 30 "neuron" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffd4ff9700 (LWP 80119)]
--Type <RET> for more, q to quit, c to continue without paging--
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb)
(gdb)
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff784a859 in __GI_abort () at abort.c:79
#2  0x00007ffff78b526e in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff79df298 "%s\n")
    at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff78bd2fc in malloc_printerr (str=str@entry=0x7ffff79e1ad8 "malloc(): unsorted double linked list corrupted") at malloc.c:5347
#4  0x00007ffff78c02ec in _int_malloc (av=av@entry=0x7fffc4000020, bytes=bytes@entry=672) at malloc.c:3744
#5  0x00007ffff78c3b95 in __libc_calloc (n=<optimized out>, elem_size=<optimized out>) at malloc.c:3428
#6  0x00007ffff7a45a1c in zlog_event_new (time_cache_count=2) at event.c:59
#7  0x00007ffff7a4c6e9 in zlog_thread_rebuild_event (a_thread=a_thread@entry=0x7fffc401b1c0, time_cache_count=<optimized out>) at thread.c:169
#8  0x00007ffff7a4fca4 in zlog (category=0x5555557e87d0,
    file=0x7ffff7ea7060 "/home/code/emq/repos/trunk/neuron/src/connection/mqtt_c_client.c", filelen=64,
    func=0x7ffff7ea74d0 <__func__.22403> "publish_callback", funclen=16, line=562, level=20,
    format=0x7ffff7ea72b2 "received publish('%s'): %s") at zlog.c:854
#9  0x00007ffff7c8351a in publish_callback (p_reconnect_state=0x7fffd8002cc0, published=0x7fffd4ff8e40)
    at /home/code/emq/repos/trunk/neuron/src/connection/mqtt_c_client.c:562
#10 0x00007ffff7e9f3a4 in __mqtt_recv () from ./libneuron-base.so
#11 0x00007ffff7e9d91c in mqtt_sync () from ./libneuron-base.so
#12 0x00007ffff7c83798 in client_refresher (context=0x7fffd8002c80) at /home/code/emq/repos/trunk/neuron/src/connection/mqtt_c_client.c:602
#13 0x00007ffff7a22609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#14 0x00007ffff7947133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) f 9
#9  0x00007ffff7c8351a in publish_callback (p_reconnect_state=0x7fffd8002cc0, published=0x7fffd4ff8e40)
    at /home/code/emq/repos/trunk/neuron/src/connection/mqtt_c_client.c:562
562             zlog_debug(client->log, "received publish('%s'): %s", topic_name,
(gdb) info args
p_reconnect_state = 0x7fffd8002cc0
published = 0x7fffd4ff8e40
(gdb) info locals
__PRETTY_FUNCTION__ = "publish_callback"
state = 0x7fffd8002d50
client = 0x7fffd8002c80
topic_name = 0x7fffc401b7c0 "/v1/nari/service/command"
__func__ = "publish_callback"
(gdb) p *published
[... 359 lines cut ...]
        if (NULL != client->log) {
            zlog_error(client->log, "topic name alloc error");
        }
        return;
    }

    memcpy(topic_name, published->topic_name, published->topic_name_size);
    topic_name[published->topic_name_size] = '\0';

    if (NULL != client->log) {
        **zlog_debug(client->log, "received publish('%s'): %s", topic_name,
                   (const char *) published->application_message);**
    }

    for (struct subscribe_tuple **p =
             (struct subscribe_tuple **) utarray_front(client->array);
         NULL != p;
         p = (struct subscribe_tuple **) utarray_next(client->array, p)) {
        if (0 == strcmp((*p)->topic, topic_name) && NULL != (*p)->handle) {
            (*p)->handle(topic_name, strlen(topic_name),
                         (void *) published->application_message,
                         published->application_message_size,
                         client->user_data);
        }
    }`

fengzeroz, 2023-01-17:
You can try to upgrade to version 2.3, the MQTT plugin has been re-implemented based on nanosdk, MQTT-C is discarded.

Jimmy-Wen, 2023-02-06:
It's the same issue as pr#826. Close it.

Jimmy-Wen, 2023-02-06:
It's the same issue as pr#826， and had been sloved.

closed 2023-02-06 by Jimmy-Wen