emqx/neuron #1076: 南向应用插件如何实现订阅数据的功能

zhouzaihang, 2023-02-21:
* Version: 2.3.6
* 场景：目前版本中MQTT是作为北向应用插件的。业务需要实现 neuron（南向订阅mqtt数据和modbus-TCP数据  -> ekuiper -> mqtt 北向应用）。
* 需求：想基于开源的mqtt北向插件改出一个mqtt南向订阅数据的插件，可以订阅特定主题下几个特定的数据；
* 问题：南向应用如何实现订阅数据的功能？  这边看到北向应用插件有一个.request ，可以调用 mqtt_plugin_request 函数，根据请求类型对应响应处理。但是这个 .request 在南向 driver 中暂未使用到。[官方文档](https://neugates.io/docs/zh/latest/sdk/modbus_example.html#driver-validate-tag)

fengzeroz, 2023-02-27:
Implement a southbound plugin that goes directly to the MQTT broker, then subscribes to the southbound MQTT plugin's group via the northbound eKuiper plugin.

zhouzaihang, 2023-02-27:
> Implement a southbound plugin that goes directly to the MQTT broker, then subscribes to the southbound MQTT plugin's group via the northbound eKuiper plugin.

是的我注意到tag的属性中有读、写和订阅，我想知道的是具体该如何实现tag的订阅属性。官方给出的ModbusTCP的实例只包含了读写的实现接口和定时读取group的实现接口，我想知道订阅tag的实现接口是哪个。或者说当plugin 监听数据的实现接口，以及监听到订阅数据后，会触发哪个接口函数。这边看到北向应用插件有一个.request ，可以调用 mqtt_plugin_request 函数，根据请求类型对应响应处理。但是这个 .request 在南向 driver 中暂未使用到。[官方文档](https://neugates.io/docs/zh/latest/sdk/modbus_example.html#request)

fengzeroz, 2023-02-28:
> 是的我注意到tag的属性中有读、写和订阅，我想知道的是具体该如何实现tag的订阅属性。官方给出的ModbusTCP的实例只包含了读写的实现接口和定时读取group的实现接口，我想知道订阅tag的实现接口是哪个。或者说当plugin 监听数据的实现接口，以及监听到订阅数据后，会触发哪个接口函数。这边看到北向应用插件有一个.request ，可以调用 mqtt_plugin_request [...]

tag的订阅属性是在核心tag缓存中处理了，这个是业务逻辑是tag变化了才进行上报到北向。我看你的意思是要去订阅mqtt topic来获取数据吧

zhouzaihang, 2023-03-01:
> tag的订阅属性是在核心tag缓存中处理了，这个是业务逻辑是tag变化了才进行上报到北向。我看你的意思是要去订阅mqtt topic来获取数据吧

是的，订阅式的数据读取，如何实现

fengzeroz, 2023-03-02:
> 是的，订阅式的数据读取，如何实现

这个本质其实去和设备读数据没啥差别，你在你的南向插件里去订阅MQTT Topic，然后想办法通过点位地址定位到从Topic中订阅得到的数据，然后通过Neuron核心的update接口把值更新出去就可以了。类似现在北向MQTT插件订阅读写请求的Topic接受请求类似。

zhouzaihang, 2023-03-04:
> 这个本质其实去和设备读数据没啥差别，你在你的南向插件里去订阅MQTT Topic，然后想办法通过点位地址定位到从Topic中订阅得到的数据，然后通过Neuron核心的update接口把值更新出去就可以了。类似现在北向MQTT插件订阅读写请求的Topic接受请求类似。

OK

closed 2023-03-04 by zhouzaihang