emqx/neuron #1081: neuron是否有必要支持系统重启

Jimmy-Wen, 2023-03-01:
您好，提个需求：
1. 现场运维人员一个习惯性的操作是，点表配置结束或者遇到配置问题时，直接重启网关设备，Dashboard是否有必要支持系统重启？

fengzeroz, 2023-03-02:
> 您好，提个需求：
> 
> 1. 现场运维人员一个习惯性的操作是，点表配置结束或者遇到配置问题时，直接重启网关设备，Dashboard是否有必要支持系统重启？（Maintenance personnel will directly reboot the gateway device at the end of configuration or if they encounter problem.）

点表配置结束重启设备应该是没必要的，Neuron的配置信息不需要重启就会生效的。
It is not necessary to reboot the device at the end of the tag table configuration, Neuron's configuration is instant and does not require a reboot.
遇到配置问题时重启设备，大概率也是无法解决问题的。
When encountering configuration problems, rebooting the device will most likely not prevent the problem.

建议对运维人员做培训，这两种情况都没必要重启设备的。
It is recommended to train the maintenance personnel, both cases are not necessary to reboot the device.

Jimmy-Wen, 2023-03-04:
了解了，多谢!

closed 2023-03-04 by Jimmy-Wen