emqx/neuron #1175: Function codes 0x06 and 0x0F not supported

evhv, 2023-04-07:
Good evening, I know this question has been asked before. Though are there any plans to support function codes 0x06 and 0x0F or possibly ETA when they will be supported?
During testing I stumbled upon couple of devices that support only fc 0x06 for setting the register, therefore all works perfect apart from writing to those devices.

fengzeroz, 2023-04-10:
Thank you for your suggestion, we already plan to support codes 0x06 and 0x0f in version 2.5.

evhv, 2023-04-10:
Thank you, thats great to know. For the time being I modified some modbus_wrap functions to work with 0x06, if needed I could clean them a bit and push to the forked repo.

hxy7yx, 2023-04-23:
Sincerely thank you for your PR. If you have any questions about Neuron, please feel free to communicate and wish you all the best.

evhv, 2023-04-23:
Thank you, that was a great learning opportunity.
All the best and wish you a good day

closed 2023-04-24 by hxy7yx