emqx/neuron #1215: file plugin read error

abchujb, 2023-04-23:
# Issue Template

Please fill in the issue as required.

* System: ubuntu linux
* Architecture: x86
* Version: 18.04
* Problem Description: read file error if set file_length = 5
* Log Information:
==9217==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x6020000040d5 at pc 0x7feee431c66e bp 0x7feed70ca920 sp 0x7feed70ca0c8
READ of size 6 at 0x6020000040d5 thread T17
    #0 0x7feee431c66d  (/usr/lib/x86_64-linux-gnu/libasan.so.4+0x5166d)
    #1 0x7feed78d56f6 in file_group_timer /home/hujb/neuron/plugins/file/file_req.c:87
    #2 0x7feed78d4af8 in driver_group_timer /home/hujb/neuron/plugins/file/file_plugin.c:149
    #3 0x55a17cedac28 in read_callback /home/hujb/neuron/src/adapter/driver/driver.c:903
    #4 0x7feee3a03a2d in event_loop /home/hujb/neuron/src/event/event_linux.c:116
    #5 0x7feee30fc6da in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76da)
    #6 0x7feee2e2561e in __clone (/lib/x86_64-linux-gnu/libc.so.6+0x12161e)

0x6020000040d5 is located 0 bytes to the right of 5-byte region [0x6020000040d0,0x6020000040d5)
allocated by thread T17 here:
    #0 0x7feee43a9d28 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded28)
    #1 0x7feed78d51b0 in file_group_timer /home/hujb/neuron/plugins/file/file_req.c:55
    #2 0x7feed78d4af8 in driver_group_timer /home/hujb/neuron/plugins/file/file_plugin.c:149
    #3 0x55a17cedac28 in read_callback /home/hujb/neuron/src/adapter/driver/driver.c:903
    #4 0x7feee3a03a2d in event_loop /home/hujb/neuron/src/event/event_linux.c:116
    #5 0x7feee30fc6da in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76da)

Thread T17 created by T0 here:
    #0 0x7feee4302d2f in __interceptor_pthread_create (/usr/lib/x86_64-linux-gnu/libasan.so.4+0x37d2f)
    #1 0x7feee3a040ef in neu_event_new /home/hujb/neuron/src/event/event_linux.c:160
    #2 0x55a17cec53c4 in neu_adapter_driver_create /home/hujb/neuron/src/adapter/driver/driver.c:136
    #3 0x55a17cea9310 in neu_adapter_create /home/hujb/neuron/src/adapter/adapter.c:106
    #4 0x55a17ce69989 in neu_mana
ger_add_node /home/hujb/neuron/src/core/manager_internal.c:78
    #5 0x55a1”

[test.txt](https://github.com/emqx/neuron/files/11303003/test.txt)

lixiumei123, 2023-04-24:
Which version of Neuron is used?

abchujb, 2023-04-24:
the main branch

lixiumei123, 2023-04-26:
The main branch has been repaired.
You can check it from [committed](https://github.com/emqx/neuron/commit/cc419307a4986676898e0b5ee66fda3796b63c9d).

closed 2023-04-28 by fengzeroz