emqx/neuron #1367: 大胆点评一下neuron

yeah1, 2023-06-30:
优点：
1、使用c语言开发，占用资源低，意味着硬件成本会低有市场竞争力，这方面比go、java的好。估计做成跨平台的问题也不大；
2、一看就是做高性能软件的开发人员写的代码，可能源于做emqx时锱铢必较的深切体会；
3、上下行框架还算完整，持续开源更新代码待观察；
缺点：
1、从论坛里的讨论及代码看通信协议感觉做的还是比较糙，当然这方面需要深刻的协议开发与现场实施调试经验积累；
2、开源版dashboard比较简陋，仍然有便利化提升的空间，如整个项目的导入导出，通过页面可选开关打开io收发通信报文查看，过滤保存等；
3、4层io与7层应用层工业通信协议未解耦分离，Modbus RTU over TCP/UDP，下次还有DLT645 over TCP/UDP呢？cjt188 over TCP/UDP呢？一般网关这部分是抽象出来的，上层使用时就是set参数 open close read write的抽象；
4、设备下弄不同组，这种在设备很多时仍然点鼠标多。可参考things cloud等都是任务模式，读写任务下是设备，设备有不同的读写点位方案，这种灵活编排读写任务模式。

总之设计思想是第一位的，地基要打牢，地基比盖几层楼重要。个人观点，请斧正！
当然在众多开源类似项目中还属不错的，应多鼓励支持、贡献改进，暂时这么多，后续再补充。

gc87, 2023-06-30:
Thank you for your endorsement and suggestions for the project, and all are welcome to continue the discussion on this issue.

Jimmy-Wen, 2023-07-02:
个人十分赞赏Neuron得代码架构及风格，也觉得Dashboard支持IO报文查看，在工程调试时确实很有必要。而当前IO报文还是通过日志文件 或者 console控制台查看。

closed 2023-07-31 by fengzeroz