emqx/neuron #1458: Licensing question, 30 tags limit on Modbus TCP

evhv, 2023-08-17:
Good afternoon, did anything change in Neuron licensing terms?
After updating to v.2.5.3, Modbus TCP plugin errors with the message that I exceed the allowed number of tags, >30. Though after reading the docs again it seems that this limit is only for commercial plugins and is not applicable to Modbus TCP, could you please clarify on that?
The same thing happens on the clean install after I add more than 30 tags. 

All the best, 
Ev

evhv, 2023-08-18:
To add more information, just rolled back to Neuron 2.5.2 and the issue disappeared, seems to be a bug in 2.5.3

fengzeroz, 2023-08-22:
Modbus RTU/TCP is packaged closed-source version in the version 2.5.3 package. If you want to use Modbus RTU/TCP without restrictions, you can compile and use the Modbus RTU/TCP plugin from the Neuron repository.

evhv, 2023-08-22:
Got it, thank you very much

closed 2023-08-23 by fengzeroz

huanghe-1, 2023-10-24:
@fengzeroz  package is commercial and requires license，but the source code is open,someone can build  theirself,why? What's the point of doing this。

QQDQ, 2023-10-24:
@huanghe-1 The Neuron commercial software downloaded from the company's official website means that the company will provide commercial technical support and services for all drivers, and all drivers will be treated equally. If you want to use the software without any restrictions, please compile and package it yourself.