emqx/neuron #1518: mqtt插件会自动退出并重新加载

jerryma0912, 2023-09-08:
# Issue Template

Please fill in the issue as required.

* System: ubuntu 18.04
* Architecture: arm64
* Version: latest
问题描述：
1. mqtt北向插件，通过读Topic：/neuron/mqtt/read/req 同时发送6条请求，偶现在未完全响应的情况下，出现mqtt plugin重新加载的问题，但在大多数情况下是正常的，即会通过响应topic：/neuron/mqtt/read/resp响应6条结果。
2. 上述问题非必现，但我在测试过程中，12小时内已经发生了3次。

以下为日志：

2023-09-07 20:00:00 [DEBUG] mqtt_client.c:253 pub [/neuron/mqtt/test/node_3, QoS0] 871 bytes
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:515 recv [/neuron/mqtt/read/req, QoS0] 83 bytes
2023-09-07 20:00:00 [NOTICE] mqtt_handle.c:244 read uuid:8161c099-4c23-4425-9c21-da3f8655f671, group:metG1_1, node:test
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:515 recv [/neuron/mqtt/read/req, QoS0] 83 bytes
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:515 recv [/neuron/mqtt/read/req, QoS0] 83 bytes
2023-09-07 20:00:00 [NOTICE] mqtt_handle.c:244 read uuid:3388976e-6240-4abc-9956-932105f92a27, group:metG1_2, node:test
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:515 recv [/neuron/mqtt/read/req, QoS0] 83 bytes
2023-09-07 20:00:00 [NOTICE] mqtt_handle.c:244 read uuid:2349d62e-8d52-4284-80a6-0c9d23b30cb2, group:metG2_2, node:test
2023-09-07 20:00:00 [NOTICE] mqtt_handle.c:244 read uuid:9abff23b-2ba2-445b-9d30-39b3375d5c61, group:metG2_1, node:test
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:515 recv [/neuron/mqtt/read/req, QoS0] 91 bytes
2023-09-07 20:00:00 [NOTICE] mqtt_handle.c:244 read uuid:f7e27e2e-49c0-4cb2-8144-b476edbb8bf9, group:metEStorage_1, node:test
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:515 recv [/neuron/mqtt/read/req, QoS0] 91 bytes
2023-09-07 20:00:00 [NOTICE] mqtt_handle.c:244 read uuid:be656de3-8518-49bb-9ba5-9966b1d1b593, group:metEStorage_2, node:test
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:253 pub [/neuron/mqtt/bms/rack6_3, QoS0] 8693 bytes
2023-09-07 20:00:00 [DEBUG] mqtt_client.c:253 pub [/neuron/mqtt/read/resp, QoS0] 649 bytes
2023-09-07 20:00:04 [NOTICE] mqtt_plugin.c:63 initialize plugin `MQTT` success
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:80 setting ssl disabled
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:362 config client-id       : neuron
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:363 config qos             : 0
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:364 config format          : format-values
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:366 config write-req-topic : /neuron/mqtt/write/req
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:367 config write-resp-topic: /neuron/mqtt/write/resp
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:369 config cache           : 0
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:370 config cache-mem-size  : 0
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:371 config cache-disk-size : 0
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:373 config cache-sync-interval : 100
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:375 config host            : 127.0.0.1
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:376 config port            : 1883
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:379 config username        : admin
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:382 config password        : ********
2023-09-07 20:00:04 [NOTICE] mqtt_config.c:386 config ssl             : 0
2023-09-07 20:00:04 [DEBUG] mqtt_client.c:1094 cache disabled
2023-09-07 20:00:04 [DEBUG] mqtt_client.c:1022 tls disabled
2023-09-07 20:00:04 [NOTICE] mqtt_plugin.c:283 config plugin `MQTT` success
2023-09-07 20:00:04 [NOTICE] mqtt_plugin.c:313 start plugin `MQTT` success
2023-09-07 20:00:04 [NOTICE] mqtt_client.c:435 mqtt client connected
2023-09-07 20:00:04 [NOTICE] mqtt_plugin.c:33 plugin `MQTT` connected

eeff, 2023-09-08:
Please provide the log from `neuron.log` with the timestamp match the problem you specified.
For example, around timestamp `2023-09-07 19:00:00`.

jerryma0912, 2023-09-08:
Can you provide an email address? I will email the log to you
version：Neuron 2.6.0-dev (d4e9e63)

eeff, 2023-09-08:
> Can you provide an email address? I will email the log to you version：Neuron 2.6.0-dev (d4e9e63)

Did you modify the code ? If so, I hardly can draw any conclusion without information about your changes.

jerryma0912, 2023-09-10:
Oh yes, your question inspired me!

I have modified one thing, that is, the response format of the read request. The modified response format is consistent with the actively reported attribute format. But I'm not sure what's wrong with what I modified, can you help me?

The modified content is as follows：

-------

In the `neu_json_mqtt.h`:   the following structure is added:
```
typedef struct {
     char* group;
     char* node;
     uint64_t timestamp;
     char * uuid;
} json_mqtt_unify_format_t;
```
------------
In the `neu_json_mqtt.c`： the following function is added:
```
int json_encode_mqtt_unify_format_resp(void *json_object, void *param)
{
    int                       ret  = 0;
    json_mqtt_unify_format_t *resp = (json_mqtt_unify_format_t *) param;
    neu_json_elem_t resp_elems[] = { {
                                     .name      = "node",
                                     .t         = NEU_JSON_STR,
                                     .v.val_str = resp->node,
                                 },
                                 {
                                     .name      = "group",
                                     .t         = NEU_JSON_STR,
                                     .v.val_str = resp->group,
                                 },
                                 {
                                     .name      = "timestamp",
                                     .t         = NEU_JSON_INT,
                                     .v.val_int = resp->timestamp,
                                 },
                                 {
                                     .name      = "uuid",
                                     .t         = NEU_JSON_STR,
                                     .v.val_str = resp->uuid,
                                 }  };
    ret = neu_json_encode_field(json_object, resp_elems,NEU_JSON_ELEM_SIZE(resp_elems));

    return ret;
}
```
-------------
In the `mqtt_handle.c`: the `generate_read_resp_json` function has been modified:
```
static char *generate_read_resp_json(neu_plugin_t *         plugin,
                                     neu_json_mqtt_t *      mqtt,
                                     neu_resp_read_group_t *data)
{
    neu_resp_tag_value_meta_t *tags     = data->tags;
    uint16_t                   len      = data->n_tag;
    char *                     json_str = NULL;

   // ------the following has been added ------
    json_mqtt_unify_format_t header   = {
        .group     = (char *) data->group,
        .node      = (char *) data->driver,
        .timestamp = global_timestamp,
        .uuid      = (char *) mqtt->uuid};
  // -----------------end-----------------

    neu_json_read_resp_t       json     = { 0 };

    if (0 != tag_values_to_json(tags, len, &json)) {
        plog_error(plugin, "tag_values_to_json fail");
        return NULL;
    }

    // ----------- the following has been modified -----------
//    neu_json_encode_with_mqtt(&json, neu_json_encode_read_resp, mqtt,
//                              neu_json_encode_mqtt_resp, &json_str);
    neu_json_encode_with_mqtt(&json, neu_json_encode_read_resp1, &header,
                          json_encode_mqtt_unify_format_resp,
                          &json_str);

    for (int i = 0; i < json.n_tag; i++) {
        if (json.tags[i].n_meta > 0) {
            free(json.tags[i].metas);
        }
    }
    // ----------------------end-------------------

    if (json.tags) {
        free(json.tags);
    }
    return json_str;
}
```

eeff, 2023-09-11:
I think as long as your code does not segfault, it does not relate to the problem reported. 
If you modify the code in neuron core, then probably it is not the mqtt plugin causing the problem.

jerryma0912, 2023-09-11:
I only modified this part of the code, the core code was not updated. So I still think that this issue has nothing to do with the content I modified.

jerryma0912, 2023-09-11:
> I only modified this part of the code, the core code was not updated. So I still think that this issue has nothing to do with the content I modified.

I suggest use the official version to test this problem  : )

closed 2025-07-31 by fengzeroz