emqx/neuron #1613: on modbus rtu ethernet server mode ,when change one south device's  config and save ,all  south device' connect reset.

huanghe-1, 2023-10-26:
correcttly, it should only affects itself who change the config ,don't affects  other device connect.

* Version:2.6.1

hxy7yx, 2023-10-26:
> correcttly, it should only affects itself who change the config ,don't affects other device connect.
> 
> * Version:2.6.1

Thank you for raising this concern. As of now, we won't be making any immediate changes. Every time the configuration is resubmitted, it is designed to reset the plugin state. Since altering the configuration isn't frequent, this approach helps us maintain a more streamlined and simplified logic. We truly value community contributions and feedback. If you'd like to propose an alternative, please feel free to submit a pull request, and we'd be happy to discuss it further.

huanghe-1, 2023-11-02:
ok,too many years don't  touch c language，i can try.

closed 2023-11-17 by hxy7yx