emqx/neuron #1664: failed to compiling the project

stilong, 2023-11-11:
[root@k8s-npu build]# cmake ..
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:56 (find_package):
  Could not find a package configuration file provided by "nng" with any of
  the following names:

    nngConfig.cmake
    nng-config.cmake

  Add the installation prefix of "nng" to CMAKE_PREFIX_PATH or set "nng_DIR"
  to a directory containing one of the above files.  If "nng" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/root/lgh/neuron/build/CMakeFiles/CMakeOutput.log".

gc87, 2023-11-11:
You need to install the nng sdk.

https://github.com/emqx/neuron/blob/main/Install-dependencies.md

stilong, 2023-11-21:
> You need to install the Nano-SDK.
> 
> https://github.com/emqx/neuron/blob/main/Install-dependencies.md

 Nano-SDK has been installed,but compiling errors ocured again

stilong, 2023-11-22:
solved

closed 2023-12-11 by gc87