emqx/neuron #1696: 交叉编译报错 fatal error: gnu/libc-version.h: No such file or directory

ZXiaoPang, 2023-11-24:
# Issue Template
是我用的交叉编译工具链没有提供GNU C的ucontext库导致的吗？

我应该怎么处理这个错误

huanghe-1, 2023-11-25:
为什么要交叉编译，你是想编译windows版本的neuron？

ZXiaoPang, 2023-11-25:
我这边有个linux的工控器， 我需要neuron 交叉编译到设备链， 交叉编译工具链是厂商提供的

ZXiaoPang, 2023-11-25:
目前交叉编译完之后  运行报这个错误

ZXiaoPang, 2023-11-25:
[image]

huanghe-1, 2023-11-26:
实在不行，不要交叉编译了，弄个虚拟机或者云服务器，直接在目标主机系统上编译，我倒是想编译成windows版本运行的，不知道可行不

fengzeroz, 2023-11-27:
> ![截屏2023-11-25 19 15 44](https://private-user-images.githubusercontent.com/13376829/285587616-4215a819-ae0c-45f8-9fd0-51447215a045.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDEwNTE0MjgsIm5iZiI6MTcwMTA1MTEyOCwicGF0aCI6Ii8xMzM3NjgyOS8yODU1ODc2MTYtNDIxNWE4MTktYWUwYy00NWY4LTlmZDAtNTE0NDcyMTVhMDQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMTI3VDAyMTIwOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1MjBkOGMxMzE2MDhhZWVjMmIxOTg3NzYwOWRlZTAwZDU2NWRiZTVmMDBjMzIzN2IwMzczZmE1ZjNmZDAyNDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.idKoDjYyqyLxX0Gd394_BNzbJ4lYm9cIuGPXD0-S2wA) 目前交叉编译完之后 运行报这个错误

可以看看对应代码，核心端口没有绑上

ZXiaoPang, 2023-11-28:
我把这个地方127.0.0.1 改成本机地址就可以正常运行了，  用127.0.0.1获取不到localhost

ZXiaoPang, 2023-11-28:
[image]
是我的hosts文件配置有问题吗？

ZXiaoPang, 2023-11-29:
已解决了  配置文件有问题  我的localhost映射的是8.8.8.8

closed 2023-11-29 by ZXiaoPang