emqx/neuron #1790: Latest Docker container (2.7.0) - 404 when navigating to IP:7000

blownupp, 2024-01-02:
# Issue Template

Please fill in the issue as required.

* System: Ubuntu Docker Host
* Architecture: amd64
* Version: 2.7.0
* Problem Description: After launching the latest Neuron Docker container, navigating to the dashboard (x.x.x.x:7000) yields a 404 Not Found error. Switching to the previous release (emqx/neuron:2.6.3) works without issue - it's only when using emqx/neuron:latest or emqx/neuron:2.7.0 that this happens.
* Log Information:

dbwaax, 2024-01-04:
我也遇到了这个问题

gc87, 2024-01-15:
Starting with Neuron 2.7 the dashboard is no longer available, the UI is integrated into NeuronEX and can be used directly with NeuronEX.

kidproquo, 2024-06-22:
Shouldn't the documentation reflect this drastic change? All the docs (quickstart, etc.) refer to pulling the latest docker image and accessing the dashboard at http://IP:7000.

StefanOMV, 2024-09-18:
Im completely new to the project and wanted to try it out so if I start the neuron docker container as described I cant access the web ui. Do I have to start another docker container to access the ui? What does "the UI is integrated into NeuronEX" mean. Is the UI no longer available without a license?

joejej, 2025-01-05:
As of today, 2.6.3 now  installs WITH the license qualifier - so thats been changed too. Yet they still claim its open source albeit with only modbus and mqtt drivers. this is like claiming a box of free crisps that you have a voucher for. But it turns up empty. customer service says ahhhhhhh you expected crisps in your box of free crisps. nah nah nah....  you have to pay for those. you got your free box, whats the problem? But we will let you munch a few crisps for two weeks. then a couple more for another two weeks...

GripenANM, 2025-01-06:
> As of today, 2.6.3 now installs WITH the license qualifier - so thats been changed too. Yet they still claim its open source albeit with only modbus and mqtt drivers. this is like claiming a box of [...]

If you really want a working UI, you can copy the `/dist` folder from the compiled version in the github release into the built version of the source code from this repo. The `neuron-dashboard` repo seems to be lacking behind and is creating some issues when I attempted to run it with the latest version of CE neuron.

Just in case if you want a `dockerfile` to build this repo and run your own docker image, here you go:

```dockerfile
FROM ubuntu:24.04 AS base

RUN apt update
RUN apt install -y libssl-dev
RUN apt install -y openssl
RUN apt install -y unzip
RUN apt install -y git

#builder
FROM base AS builder

RUN apt install -y build-essential
RUN apt install -y cmake
RUN apt install -y ninja-build
RUN apt install -y curl
RUN apt install -y wget
RUN apt install -y autoconf
RUN apt install -y libtool

WORKDIR /opt

RUN git clone -b 1.2.15 https://github.com/HardySimpson/zlog.git
RUN git clone https://github.com/neugates/jansson.git
RUN git clone -b v2.16.12 https://github.com/Mbed-TLS/mbedtls.git
RUN git clone -b neuron https://github.com/neugates/NanoSDK.git
RUN git clone -b v1.13.1 https://github.com/benmcollins/libjwt.git
RUN git clone -b release-1.11.0 https://github.com/google/googletest.git
RUN curl -o sqlite3.tar.gz https://www.sqlite.org/2022/sqlite-autoconf-3390000.tar.gz
RUN wget --no-check-certificate --content-disposition https://github.com/protocolbuffers/protobuf/releases/download/v3.20.1/protobuf-cpp-3.20.1.tar.gz
RUN git clone -b v1.4.0 https://github.com/protobuf-c/protobuf-c.git
RUN git clone -b 2.10.2 https://github.com/emqx/neuron.git

RUN cd zlog && make && make install
RUN cd jansson && mkdir build && cd build && cmake -DJANSSON_BUILD_DOCS=OFF -DJANSSON_EXAMPLES=OFF .. && make && make install
RUN cd mbedtls && mkdir build && cd build && cmake -DUSE_SHARED_MBEDTLS_LIBRARY=OFF -DENABLE_TESTING=OFF -DCMAKE_POSITION_INDEPENDENT_CODE=ON .. && make && make install
RUN cd NanoSDK && mkdir build && cd build && cmake -DBUILD_SHARED_LIBS=OFF -DNNG_TESTS=OFF -DNNG_ENABLE_SQLITE=ON -DNNG_ENABLE_TLS=ON .. && make && make install
RUN cd libjwt && mkdir build && cd build && cmake -DENABLE_PIC=ON -DBUILD_SHARED_LIBS=OFF .. && make && make install
RUN cd googletest && mkdir build && cd build && cmake .. && make && make install
RUN mkdir sqlite3 && tar xzf sqlite3.tar.gz --strip-components=1 -C sqlite3 && cd sqlite3 && ./configure CFLAGS=-fPIC && make && make install
RUN tar -xzvf protobuf-cpp-3.20.1.tar.gz && cd protobuf-3.20.1 && ./configure --enable-shared=no CFLAGS=-fPIC CXXFLAGS=-fPIC && make && make install
RUN cd protobuf-c && ./autogen.sh && ./configure  --disable-protoc --enable-shared=no CFLAGS=-fPIC CXXFLAGS=-fPIC && make && make install
RUN cd neuron && mkdir build && cd build && cmake -DCMAKE_BUILD_TYPE=Release -DDISABLE_WERROR=1 .. && make

#runner
FROM base AS runner

RUN apt install -y wget

COPY --from=builder /usr/lib/x86_64-linux-gnu/libasan.so.8.0.0 /usr/lib/x86_64-linux-gnu/
COPY --from=builder /usr/lib/x86_64-linux-gnu/libasan.so.8 /usr/lib/x86_64-linux-gnu/
COPY --from=builder /usr/local/lib/ /usr/local/lib/
COPY --from=builder /usr/local/bin/ /usr/local/bin/

COPY --from=builder /opt/neuron/build/config /opt/neuron/config
COPY --from=builder /opt/neuron/build/libneuron-base.so /opt/neuron
COPY --from=builder /opt/neuron/build/logs /opt/neuron/logs
COPY --from=builder /opt/neuron/build/neuron /opt/neuron
COPY --from=builder /opt/neuron/build/persistence /opt/neuron/persistence
COPY --from=builder /opt/neuron/build/plugins /opt/neuron/plugins
COPY --from=builder /opt/neuron/build/simulator /opt/neuron/simulator
COPY --from=builder /opt/neuron/build/tests /opt/neuron/tests

WORKDIR /opt/neuron

COPY ./dist ./dist
#RUN wget https://github.com/emqx/neuron-dashboard/releases/download/2.4.9/neuron-dashboard-lite.zip
#RUN unzip neuron-dashboard-lite.zip

EXPOSE 7000

CMD ["./neuron"]
```
> [!NOTE]
> If you want UI, download the files from the latest github release, and copy  the `/dist` folder to the directory you are running the docker build command.
>
> Your folder should be something like this:
> ```
> /dist
> dockerfile
> ```

joejej, 2025-01-06:
Thanks - yes I actually just did a reinstall following the github instructions. installed emqx/neuron 2.10.2 deb. Then out of interest downloaded, and unzipped 2.4.9 neuron-dashboard - which unzipped into its own dist directory. The webpage opened straight up. No extra build or config needed. Then downloaded dashboard 2.6.3 - which was just source code. 2.4.9 still had the license limit contrary to the open source claim. 2.6.3 - im not going to bother doing a build. emqx has clearly made a commercial decision, and im not going to use a product on my own home that i could end up having to pay $300+ per month for....  I still have the complete 2.5.3 deb i can roll back to and works for me....

But whats this CE neuron you mention? An earlier fork? Just accessed cli? Im not having any success googling it up. :D with a name like neuron Im starting to feel like im trying to find a brain lab!

GripenANM, 2025-01-06:
> Thanks - yes I actually just did a reinstall following the github instructions. installed emqx/neuron 2.10.2 deb. Then out of interest downloaded, and unzipped 2.4.9 neuron-dashboard - which [...]

Apparently, the files from the GitHub release and the official website are not the actually built directly from the code in this repo; They did some post processing in their shell script. 
To actually use the "open source license" and get away with the tag and node limits, you will need to **Build From Source**. See #2351 for response from the repo maintainer. 

> The CE neuron is just the version that you build from source, sorry for the confusion.

wonka929, 2025-06-18:
Hi everyone.
Just to understand if I'm correct: basically 2.7 is the dead end for Neuron as open source project right?

In case, why don't we fork it?

fengzeroz, 2025-08-27:
The latest image maintained for Neuron is `emqx/neuron:2.6.9`, and images for versions `2.7.x` and later are no longer provided. You can refer to the readme and use the older dashboard.