emqx/neuron #1915: compile error

wzf2001, 2024-03-21:
[zf@zf-virtual-machine /home/zf]$git clone https://github.com/emqx/neuron
正克隆到 'neuron'...
remote: Enumerating objects: 25938, done.
remote: Counting objects: 100% (194/194), done.
remote: Compressing objects: 100% (124/124), done.
remote: Total 25938 (delta 81), reused 143 (delta 66), pack-reused 25744
接收对象中: 100% (25938/25938), 7.67 MiB | 2.05 MiB/s, 完成.
处理 delta 中: 100% (17972/17972), 完成.
[zf@zf-virtual-machine /home/zf]$cd neuron/
[zf@zf-virtual-machine /home/zf/neuron]$mkdir build && cd build
[zf@zf-virtual-machine /home/zf/neuron/build]$cmake ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zf/neuron/build
[zf@zf-virtual-machine /home/zf/neuron/build]$make
[  0%] Built target neuron-version
[  1%] Building C object CMakeFiles/neuron-base.dir/src/base/tag.c.o
[  2%] Building C object CMakeFiles/neuron-base.dir/src/base/neu_plugin_common.c.o
[  3%] Building C object CMakeFiles/neuron-base.dir/src/base/tag_sort.c.o
[  4%] Building C object CMakeFiles/neuron-base.dir/src/base/group.c.o
[  4%] Building C object CMakeFiles/neuron-base.dir/src/base/metrics.c.o
[  5%] Building C object CMakeFiles/neuron-base.dir/src/base/msg.c.o
[  6%] Building C object CMakeFiles/neuron-base.dir/src/connection/connection.c.o
[  7%] Building C object CMakeFiles/neuron-base.dir/src/connection/connection_eth.c.o
[  8%] Building C object CMakeFiles/neuron-base.dir/src/connection/mqtt_client.c.o
[  8%] Building C object CMakeFiles/neuron-base.dir/src/event/event_linux.c.o
[  9%] Building C object CMakeFiles/neuron-base.dir/src/event/event_unix.c.o
[ 10%] Building C object CMakeFiles/neuron-base.dir/src/utils/asprintf.c.o
[ 11%] Building C object CMakeFiles/neuron-base.dir/src/utils/json.c.o
[ 12%] Building C object CMakeFiles/neuron-base.dir/src/utils/http.c.o
[ 12%] Building C object CMakeFiles/neuron-base.dir/src/utils/http_handler.c.o
[ 13%] Building C object CMakeFiles/neuron-base.dir/src/utils/neu_jwt.c.o
[ 14%] Building C object CMakeFiles/neuron-base.dir/src/utils/base64.c.o
[ 15%] Building C object CMakeFiles/neuron-base.dir/src/utils/async_queue.c.o
[ 16%] Building C object CMakeFiles/neuron-base.dir/src/utils/log.c.o
[ 16%] Building C object CMakeFiles/neuron-base.dir/src/persist/persist.c.o
[ 17%] Building C object CMakeFiles/neuron-base.dir/src/persist/sqlite.c.o
[ 18%] Building C object CMakeFiles/neuron-base.dir/src/persist/json/persist_json_plugin.c.o
[ 19%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_error.c.o
[ 20%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_fn.c.o
[ 20%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_global_config.c.o
[ 21%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_group_config.c.o
[ 22%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_log.c.o
[ 23%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_login.c.o
[ 24%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_mqtt.c.o
[ 24%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_node.c.o
[ 25%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_plugin.c.o
[ 26%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_rw.c.o
[ 27%] Building C object CMakeFiles/neuron-base.dir/src/parser/neu_json_tag.c.o
[ 28%] Linking C shared library libneuron-base.so
[ 28%] Built target neuron-base
[ 29%] Building C object CMakeFiles/neuron.dir/src/main.c.o
[ 30%] Building C object CMakeFiles/neuron.dir/src/argparse.c.o
[ 30%] Building C object CMakeFiles/neuron.dir/src/daemon.c.o
[ 31%] Building C object CMakeFiles/neuron.dir/src/core/manager_internal.c.o
[ 32%] Building C object CMakeFiles/neuron.dir/src/core/manager.c.o
[ 33%] Building C object CMakeFiles/neuron.dir/src/core/subscribe.c.o
[ 34%] Building C object CMakeFiles/neuron.dir/src/core/plugin_manager.c.o
[ 34%] Building C object CMakeFiles/neuron.dir/src/core/node_manager.c.o
[ 35%] Building C object CMakeFiles/neuron.dir/src/core/storage.c.o
[ 36%] Building C object CMakeFiles/neuron.dir/src/adapter/msg_q.c.o
[ 37%] Building C object CMakeFiles/neuron.dir/src/adapter/storage.c.o
[ 38%] Building C object CMakeFiles/neuron.dir/src/adapter/adapter.c.o
[ 38%] Building C object CMakeFiles/neuron.dir/src/adapter/driver/cache.c.o
[ 39%] Building C object CMakeFiles/neuron.dir/src/adapter/driver/driver.c.o
[ 40%] Building C object CMakeFiles/neuron.dir/plugins/restful/handle.c.o
[ 41%] Building C object CMakeFiles/neuron.dir/plugins/restful/log_handle.c.o
[ 42%] Building C object CMakeFiles/neuron.dir/plugins/restful/metric_handle.c.o
[ 42%] Building C object CMakeFiles/neuron.dir/plugins/restful/normal_handle.c.o
[ 43%] Building C object CMakeFiles/neuron.dir/plugins/restful/rw_handle.c.o
[ 44%] Building C object CMakeFiles/neuron.dir/plugins/restful/adapter_handle.c.o
[ 45%] Building C object CMakeFiles/neuron.dir/plugins/restful/datatag_handle.c.o
[ 46%] Building C object CMakeFiles/neuron.dir/plugins/restful/global_config_handle.c.o
[ 46%] Building C object CMakeFiles/neuron.dir/plugins/restful/group_config_handle.c.o
[ 47%] Building C object CMakeFiles/neuron.dir/plugins/restful/plugin_handle.c.o
[ 48%] Building C object CMakeFiles/neuron.dir/plugins/restful/version_handle.c.o
[ 49%] Building C object CMakeFiles/neuron.dir/plugins/restful/rest.c.o
[ 50%] Building C object CMakeFiles/neuron.dir/plugins/restful/user.c.o
[ 50%] Linking C executable neuron
/usr/bin/ld: libneuron-base.so: undefined reference to `json_realp'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/neuron.dir/build.make:514：neuron] 错误 1
make[1]: *** [CMakeFiles/Makefile2:309：CMakeFiles/neuron.dir/all] 错误 2
make: *** [Makefile:146：all] 错误 2

hxy7yx, 2024-03-21:
Check the installation path of jansson.

wzf2001, 2024-03-21:
where it should be？
I install jansson and neuron are in the same directory

hxy7yx, 2024-03-21:
If the installation was performed using sudo? Also, could you verify if jansson has been installed in /usr/local?

wzf2001, 2024-03-21:
this is my usr/local/bin:
[zf@zf-virtual-machine /usr/local/bin]$ls
benchmark   crypt_and_hash  dtls_server          gen_random_havege  mpi_demo    pk_sign                    rsa_encrypt  sqlite3          ssl_pthread_server  zlog-chk-conf
cert_app    dh_client       gen_entropy          hello              nngcat      pk_verify                  rsa_genkey   ssl_client1      ssl_server
cert_req    dh_genprime     generic_sum          key_app            pem2der     query_compile_time_config  rsa_sign     ssl_client2      ssl_server2
cert_write  dh_server       gen_key              load_roots         pk_decrypt  req_app                    rsa_verify   ssl_fork_server  strerror
crl_app     dtls_client     gen_random_ctr_drbg  mini_client        pk_encrypt  rsa_decrypt                selftest     ssl_mail_client  udp_proxy
usr/local/lib:cmake            libgtest.a       libjansson.la    libjansson.so.4.14.0  libmbedtls.a   libsqlite3.a   libsqlite3.so.0      libzlog.so      pkgconfig
libgmock.a       libgtest_main.a  libjansson.so    libjwt.a              libmbedx509.a  libsqlite3.la  libsqlite3.so.0.8.6  libzlog.so.1    python3.10
libgmock_main.a  libjansson.a     libjansson.so.4  libmbedcrypto.a       libnng.a       libsqlite3.so  libzlog.a            libzlog.so.1.2
and I into the jansson/build run **make check**, it generate error:
The following tests FAILED:
	 28 - encoding-flags__real-precision (Failed)
	 60 - valid__real-exponent (Failed)
	 61 - valid__real-exponent__strip (Failed)
	 68 - valid__real-subnormal-number (Failed)
	 69 - valid__real-subnormal-number__strip (Failed)
LastTest.log msg:
Test Failed.
"encoding-flags__real-precision" end time: Mar 21 10:35 CST
"encoding-flags__real-precision" time elapsed: 00:00:00
----------------------------------------------------------

29/213 Testing: encoding-flags__sort-keys
29/213 Test: encoding-flags__sort-keys
Command: "/home/zf/jansson/build/bin/json_process" "/home/zf/jansson/test/suites/encoding-flags/sort-keys"
Directory: /home/zf/jansson/build
"encoding-flags__sort-keys" start time: Mar 21 10:35 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.00 sec
----------------------------------------------------------

Test Failed.
"valid__real-exponent" end time: Mar 21 10:35 CST
"valid__real-exponent" time elapsed: 00:00:00
----------------------------------------------------------

61/213 Testing: valid__real-exponent__strip
61/213 Test: valid__real-exponent__strip
Command: "/home/zf/jansson/build/bin/json_process" "--strip" "/home/zf/jansson/test/suites/valid/real-exponent"
Directory: /home/zf/jansson/build
"valid__real-exponent__strip" start time: Mar 21 10:35 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.00 sec
----------------------------------------------------------
Test Failed.
"valid__real-exponent__strip" end time: Mar 21 10:35 CST
"valid__real-exponent__strip" time elapsed: 00:00:00
----------------------------------------------------------

62/213 Testing: valid__real-fraction-exponent
62/213 Test: valid__real-fraction-exponent
Command: "/home/zf/jansson/build/bin/json_process" "/home/zf/jansson/test/suites/valid/real-fraction-exponent"
Directory: /home/zf/jansson/build
"valid__real-fraction-exponent" start time: Mar 21 10:35 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.00 sec
----------------------------------------------------------

Test Failed.
"valid__real-subnormal-number__strip" end time: Mar 21 10:35 CST
"valid__real-subnormal-number__strip" time elapsed: 00:00:00
----------------------------------------------------------

70/213 Testing: valid__real-underflow
70/213 Test: valid__real-underflow
Command: "/home/zf/jansson/build/bin/json_process" "/home/zf/jansson/test/suites/valid/real-underflow"
Directory: /home/zf/jansson/build
"valid__real-underflow" start time: Mar 21 10:35 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.00 sec
----------------------------------------------------------

Test Failed.
"valid__real-subnormal-number__strip" end time: Mar 21 10:35 CST
"valid__real-subnormal-number__strip" time elapsed: 00:00:00
----------------------------------------------------------

70/213 Testing: valid__real-underflow
70/213 Test: valid__real-underflow
Command: "/home/zf/jansson/build/bin/json_process" "/home/zf/jansson/test/suites/valid/real-underflow"
Directory: /home/zf/jansson/build
"valid__real-underflow" start time: Mar 21 10:35 CST
Output:
----------------------------------------------------------
<end of output>
Test time =   0.00 sec
----------------------------------------------------------

hxy7yx, 2024-03-21:
So, please check the issue until the dependencies are successfully installed correctly before installing Neuron.

wzf2001, 2024-03-22:
Yes,you are right,I del all related to jansson  file or lib in /usr/local and  reinstall jansson then the library was successfully installed.

closed 2024-03-22 by hxy7yx