emqx/neuron #2065: How to send MQTT at internals less than 100ms?

leo-leonardo, 2024-06-04:
# How to send MQTT collection message less than 100ms internal?

In neuron-dashboard, I try to increase the frequency sending MQTT message of South-Devices to MQTT broker. But when I set the `intenal=20ms`, an error message is shown to tell me the minimum of internal is 100.

Please tell me how to set the internal or where (or which file) I can modify the source code. Any tip might be helpful.

* System: Ubuntu 20.04

[image]

fengzeroz, 2024-06-05:
The release version of the data frequency is limited to 100ms, you can modify the code to remove this limitation.

leo-leonardo, 2024-06-05:
> The release version of the data frequency is limited to 100ms, you can modify the code to remove this limitation.

Thanks for your clue! I found the config value at `mqtt_client.h`.

closed 2024-06-05 by leo-leonardo