emqx/neuron #2115: The MQTT write operation on South Device is invalid.

leo-leonardo, 2024-06-20:
# The MQTT write operation on South Device is invalid.

When I try to overwrite value on my tag, I send the MQTT message to the neuron write topic. But, the message seems not received by neuron, the value is not write to the tag.
[image]

The topic name is checked.
[image]

And the group and tag are also checked.
[image]

The subscription data is captured correctly.
[image]

**Did I miss any configuration? I have controlled successfully in the `apt` installed version, but this trouble happened in the compiled version.**

eeff, 2024-06-21:
@leo-leonardo make sure the broker which the mqtt plugin connects is exactly the broker you publish the write requests. If that's the case and you are still not able to write the tag, please check the log of the mqtt plugin to find any clue of problem.

leo-leonardo, 2024-06-21:
> @leo-leonardo make sure the broker which the mqtt plugin connects to is exactly the broker you publish the write requests. If that's the case and you are still not able to write the tag, please check the log of the mqtt plugin to find any clue of problem.

Thank you! I check the log and find that Neuron can not decode the MQTT message for the `'` I use. Now, `mosquitto_pub -t /topic -m '{"key":"value"}'` is success.

closed 2024-06-21 by leo-leonardo