emqx/neuron #2127: Modbus TCP always requests address 0

dakhnod, 2024-07-01:
I have the following configuration for a Modbus TCP tag:
[image]

And this is the log produced:
```
2024-07-01 21:13:23:747 [NOTICE] [Modbus sim] modbus_tcp.c:252 validate tag success, name: uint16, address: 2!30001, type: 4, slave id: 2, start address: 0, n register: 1, area: input register
2024-07-01 21:13:23:747 [NOTICE] [Modbus sim] modbus_tcp.c:252 validate tag success, name: uint16, address: 2!30001, type: 4, slave id: 2, start address: 0, n register: 1, area: input register
2024-07-01 21:13:23:747 [NOTICE] [neuron] http.c:55 <0x786e7c001cf0> PUT /api/v2/tags [200]
2024-07-01 21:13:23:783 [NOTICE] [neuron] http.c:55 <0x786e7c001cf0> GET /api/v2/tags?node=Modbus+sim&group=registers&name= [200]
2024-07-01 21:13:24:060 [NOTICE] [neuron] driver.c:1639 group: registers changed, timestamp: 1719839603747286
2024-07-01 21:13:24:061 [ERROR] [Modbus sim] modbus_req.c:264 modbus device response error, skip, 2!0
2024-07-01 21:13:25:081 [ERROR] [Modbus sim] modbus_req.c:264 modbus device response error, skip, 2!0
2024-07-01 21:13:26:101 [ERROR] [Modbus sim] modbus_req.c:264 modbus device response error, skip, 2!0
```
As can be seen, the start address registers as 0 and 0 is always requested. How can I make neuron request the actual register address?

dakhnod, 2024-07-01:
Okay, I got an idea. It seems like you cannot just request any type of data from any address.

Could you explain the meaning of the first character in the register address? E.g. 4 in 1!400001

hxy7yx, 2024-07-02:
The first character represents the storage area.

`typedef enum modbus_area {
    MODBUS_AREA_COIL           = 0,
    MODBUS_AREA_INPUT          = 1,
    MODBUS_AREA_INPUT_REGISTER = 3,
    MODBUS_AREA_HOLD_REGISTER  = 4,
} modbus_area_e;`

closed 2024-07-10 by hxy7yx