emqx/neuron #2165: Incorrect Truncation Issue of Broadcast Message

catch6, 2024-07-25:
# Issue Template

Please fill in the issue as required.

* System: RaspberryPi OS
* Architecture:  arm64
* Version: 2.9.0

Can the problem of incorrect truncation of the length of this broadcast message be fixed?

The complete message of the FD broadcast is 0xFD 0x10 0x00 0x14 0x00 0x01 0x02 0x00 0x02 0x75 0x81, among which:

FD: Represents the broadcast message.
0x10: Function code. 0x10 indicates the function code for writing multiple holding registers.
0x00 0x14: Start address, indicating the start register address where data is to be written. Here, it indicates that the start address is 0x14 (20 in decimal).
0x00 0x01: Number of registers, indicating the number of consecutive registers to be written. Here, it indicates that 1 register is to be written.
0x02 0x00: Data to be written, the specific value to be written in the register. Here, it indicates that the value to be written is 0x0002.
0x75 0x81: CRC check, used to verify the integrity and accuracy of the message.
However, Neuron truncated from 0x01 0x02, resulting in misalignment of subsequent messages.

[image]
[image]

hxy7yx, 2024-07-26:
???I have already replied to your previous issue and fixed it in this commit #2155

catch6, 2024-07-26:
ok! thank you

closed 2024-07-26 by catch6