emqx/neuron #2190: How to stop neuron gracefully?

bulabulabula818, 2024-08-22:
# Issue Template

Please fill in the issue as required.

* System: ubuntu 20.04
* Architecture: x86_64
* Version: lasted release 
* Problem Description: I know it can start by "$ cd build $ ./neuron", but i dont now how to stop it, when i type"ctrl + c", it crashed！
* Log Information: ...

hxy7yx, 2024-08-23:
Since neuron is generally not intended to be stopped in typical use cases, there might not be a graceful way to exit the program in some situations. You can help by providing relevant logs when you encounter issues, or use the logs from problematic exits to optimize the program.

bulabulabula818, 2024-08-23:
emm, thanks for your answer, now i know it, may be it's neuron's design.

closed 2024-08-23 by bulabulabula818