emqx/neuron #2192: Mqtt plugin message payload error 3017

nick0tine, 2024-08-27:
I have created a northbound app and it has a group having 9 tags in it.

All the subscription to southbound apps is there.
Issue:
When trying to publish message to mqtt broker address of north app, it throws mqtt payload error 3017.

The above case is only when I publish message where multiple tags are passed inside an array to tags attribute of message dict.

Although if i try to publish a message with single tag without any array, it works perfectly fine.

PS: I am using neuronEX.

nick0tine, 2024-08-27:
To add, the app configuration upload format is set to tags format.

fengzeroz, 2024-08-28:
3017 meas that the driver used for writing multiple tags does not yet support writing multiple tags in a single message

nick0tine, 2024-08-29:
> 3017 meas that the driver used for writing multiple tags does not yet support writing multiple tags in a single message

By driver you mean the plugin we're using in southbound devices ? 
Or is the protocol version? 

I'm using Allen-Bradley-ControlLogix 5500 plugin.

I'm new to this, so pardon for my silly doubts.

fengzeroz, 2024-08-30:
> By driver you mean the plugin we're using in southbound devices ? Or is the protocol version?
>
> I'm using Allen-Bradley-ControlLogix 5500 plugin.
>
> I'm new to this, so pardon for my silly doubts.

Yes, Allen-Bradley 5500 plugin does not currently support writing multiple tags in a single message, future version will support.

nick0tine, 2024-09-06:
Thanks for the comments and help

closed 2024-09-06 by nick0tine