emqx/neuron #2196: How to fetch data in PLC address like %MW110.0 through ModbusTcp?

leo-leonardo, 2024-08-29:
# Issue Template

In my plc code demo, I have some variables:
[image]

How can I get data when I only have these address? I tried 
[image]
and 
[image]
but failed.

How can I transform the %MW address to integer address like 48960?

fengzeroz, 2024-08-30:
What type of PLC is being used? Different models may have different rules for mapping PLC address to Modbus address.

leo-leonardo, 2024-09-04:
> What type of PLC is being used? Different models may have different rules for mapping PLC address to Modbus address.

I am using Schneider Modicon TM172PBG18R PLC. Can I use this mapping rule to get Modbus address in other PLCs?

fengzeroz, 2024-09-05:
> > What type of PLC is being used? Different models may have different rules for mapping PLC address to Modbus address.
> 
> I am using Schneider Modicon TM172PBG18R PLC. Can I use this mapping rule to get Modbus address in other PLCs?

Probably not. You should configure the tag address according to the documentation for this plc.

leo-leonardo, 2024-09-05:
> Probably not. You should configure the tag address according to the documentation for this plc.

closed 2024-09-05 by leo-leonardo