emqx/neuron #2333: neu_plugin_op 高频次写入会导致内存泄漏。

a-jiang, 2024-12-02:
# Issue Template

我通过定时器开启一个写入操作，top 观察RES内存增长情况，发现半个小时左右内存有增长。
运行41分钟：
[image]
运行2个小时后：
[image]
write_test方法：
[image]
调用neu_plugin_op 的方法：
[image]

* System:
* Architecture:
* Version:2.7
* Problem Description:
* Log Information:

ZhouBox, 2024-12-03:
The Northbound app needs to implement the `request` interface to handle core messages. The `ctx` object for Northbound commands needs to be managed by the Northbound app itself.

a-jiang, 2024-12-04:
> The Northbound app needs to implement the `request` interface to handle core messages. The `ctx` object for Northbound commands needs to be managed by the Northbound app itself.

Thanks for the reply. I set `ctx` to NULL and the memory has stabilized, but I keep running it for 12 hours and the memory will grow.
Adjusted code:
[image]
Memory change ( about 1 hour):
[image]
Memory change ( about 12 hour):

closed 2024-12-04 by ZhouBox

a-jiang, 2024-12-09:
> The Northbound app needs to implement the `request` interface to handle core messages. The `ctx` object for Northbound commands needs to be managed by the Northbound app itself.
我这边把ctx 调整后的代码如下：
[image]
写入频次调整为100ms 。
经过测试20分钟后，内存持续增长：
[image]
27分钟后：
[image]