emqx/neuron #2501: A new “Monitor” was created, with case sensitivity, but neither can be deleted.

riklionline, 2025-06-17:
Change configuration information from the backend, or delete using administrator privileges.

gc87, 2025-06-17:
@riklionline Could you please provide the version information and the runtime logs? We need to do some analysis.

riklionline, 2025-06-18:
The neuron version is "2.6.9".

This issue is easy to reproduce. in the North Apps the system comes with a “monitor” that starts with a lowercase letter, and I created another "Monitor" that starts with an uppercase letter. These two northbound data applications can run simultaneously, but I cannot delete or rename them.

riklionline, 2025-06-18:
> [@riklionline](https://github.com/riklionline) Could you please provide the version information and the runtime logs? We need to do some analysis.

The neuron version is "2.6.9".

This issue is easy to reproduce. in the North Apps the system comes with a “monitor” that starts with a lowercase letter, and I created another "Monitor" that starts with an uppercase letter. These two northbound data applications can run simultaneously, but I cannot delete or rename them.

riklionline, 2025-06-18:
> [@riklionline](https://github.com/riklionline) Could you please provide the version information and the runtime logs? We need to do some analysis.

The neuron version is "2.6.9".

This issue is easy to reproduce. in the North Apps the system comes with a "monitor" that starts with a lowercase letter, and I created another "Monitor" that starts with an uppercase letter. These two northbound data applications can run simultaneously, but I cannot delete or rename them.