emqx/neuron #2559: 几处write tag 内存泄漏问题

liuchenn, 2025-09-03:
mqtt_handle.c 里面，在neu_json_decode_write_free调用之前已经将几个变量赋值为NULL

[image]

[image]

rw_handle.c 里handle_write函数调用neu_plugin_op发送给南向设备时，neu_msg_new函数对cmd是深度拷贝，无论neu_plugin_op返回结果如何都应该调用neu_req_write_tag_fini释放内存吧
[image]

[image]

[image]

hxy7yx, 2025-09-09:
There is no bug in the code.

1. Ownership Transfer: Setting pointers to NULL is intentional design.
2. Note that the contents of cmd copied by memcpy are pointers

hxy7yx, 2025-09-09:
You still don't understand the ownership management. There is only one logical owner of the memory at any time, and the cleanup is guaranteed regardless of success/failure.

liuchenn, 2025-09-09:
是我没注意send和recv指向的都是指针，这块代码没有内存泄漏

closed 2025-09-09 by hxy7yx